fieldOfView/Cura-OctoPrintPlugin #264: Cura on macOS gets unusably slow when using local printer discovery

lekv, 2022-01-08:
After not printing from macOS for a few months, I recently updated Cura to 4.12.1 and noticed that local printer discovery in the OctoPrint plugin settings makes the rest of Cura almost unusably slow. In particular after slicing a model, panning, zooming, and scrubbing through the layers are barely working.

Turning off local printer discovery and adding the printer manually makes things fast again.

This is happening on a somewhat dated Macbook air which otherwise works fine (1.7 GHz Core i7, 8GB RAM) without any other significant load.

Not sure if there's anything to do here but thought I'd file an issue in case anyone else is hitting this. Maybe the zeroconf update in https://github.com/fieldOfView/Cura-OctoPrintPlugin/commit/9956a56dc7b6355c6172857db9d08f3b3e85ba4f is related?

Please let me know if you'd like me to capture or upload logs, traces, etc.

fieldOfView, 2022-01-09:
Thanks for the report. Updating zeroconf is something I did hoping to fix issues for some, but it can also negatively effect things with certain network setups.

I actually did not update the zeroconf module as far as I meant to. If you are willing to experiment, please try the following:
* Download & extract https://github.com/jstasiak/python-zeroconf/archive/refs/tags/0.38.1.zip
* In Cura go to Help -> Show configuration folder
* Replace `plugins/OctoPrintPlugin/OctoPrintPlugin/python-zeroconf` with the downloaded files
* Restart Cura
* Post your Cura.log (which is also found in the configuration folder)

fieldOfView, 2022-01-21:
@lekv, did you have time to try the steps I outlined above?

fieldOfView, 2022-02-13:
Closing due to lack of response.

closed 2022-02-13 by fieldOfView