fieldOfView/Cura-OctoPrintPlugin #267: Cura - OctoPrint 404 Not Found Error

ScottoB1, 2022-03-07:
My issue is similar to #256 but with a few more issues:
Using Connect to OctoPrint 3.5.16, Windows 10 Home 64 bit
Get 404 Not Found Error
No Refresh or Verify button in Cura Connect to OctoPrint window
API key not valid. -  Cura will not save the new API Key from OctoPrint when pasted in.
Cura will not Uninstall/Reinstall/Update Connect To OctoPrint Plugin
Edit button not available in Connect to OctoPrint window for the printer instance
Tried the ipaddress/port/apikey command access line in FireFox and got a screen with tons of text (similar output to issue #256 )
What I've tried so far:
Upgrade to Cura 4.13.1 (from 4.12.1)
Manually enter API Key via Help-Show Configuration Folder-Machine_instances-metadata area
Set User APP DATA folder to non-archive and not-read only.
I've loaded 4.13.1 on my Mac, configured my two printers, and no issues so far.

It's almost to the point where doing an full uninstall of Cura followed with a fresh install of Cura might be in my near future.

Any advice and/or help is GREATLY APPRECIATED!

Thanks!

fieldOfView, 2022-03-07:
I am going to need a link to your (full) Cura.log, which you can find via Help -> Show configuration folder,

ScottoB1, 2022-03-07:
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/8201317/cura.log)

Let me know if there's anything else you need!
Thanks!

fieldOfView, 2022-03-08:
> Tried the ipaddress/port/apikey command access line in FireFox and got a screen with tons of text

I am going to have to see that "ton of text".

Tell me a bit more about your OctoPrint installation. Is it OctoPi running on a raspberry pi, or something else? Do you normally access it via port 80 or another port?

ScottoB1, 2022-03-09:
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/8201317/cura.log)

Let me know if there's anything else you need!
Thanks!
[settings.zip](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/8211150/settings.zip)

Running on one raspberry pi 3B+ (one pi for each printer, one Artillery Sidewinder X1 V4 and one Creality Ender 5 Plus).  I've been running Octoprint on the X1 for a couple of years with not software issues.  Print quality has been my issue (until now) with the X1, but most of that is this amateur's learning curve. I was trying to get my webcam (Logitech C615) running with the Ender 5+.  I've had the same camera on the X1 for about a year without any issues.  That's why I decided to move it to the Ender. I did have to go into the octoprint file and add the web cam to the Ender, could that have been when I had an opportunity to break something?  Everything was working great until last Saturday afternoon.
As an afterthought, I've included the octopi.txt file from the X1's pi.  

Thanks!
[octopi.txt](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/8211273/octopi.txt)

fieldOfView, 2022-03-09:
I still have no idea what you meant by the following:

> Tried the ipaddress/port/apikey command access line in FireFox and got a screen with tons of text

Did you open a url in Firefox? What URL? (sans apikey, you want to keep that private)
What text did you get there?

Do you get an "404 page not found" error when you open the following url? http://192.168.1.4:80/api/printer
Is 192.168.1.4 pointing to (one of your) your OctoPrint installs?

ScottoB1, 2022-03-10:
Let me see if I can straighten my cryptic messages…

 

First 192.168.1.4 is the IP address for my Artillery Sidewinder X1 printer.  It’s been on Octoprint the longest of the two printers.  I’m trying to fix one printer at a time so I don’t inadvertently “break” both printers.

 

The command I typed in Firefox to get the text in the json file was:

http://192.168.1.4:80/api/settings?apikey=my current API key for Octoprint

 

It took less than a second to open the .json file.  I thought I posted the file (settings.zip) to thread in Github.  Sorry, it was a little late last night by the time I got to my emails if it didn’t post.  I also included the octopi.txt file in case I might have phat phingered something in the file when I was adding the web cam.

 

I’m not opposed to uninstalling Cura and the Octoprint plugin and doing a fresh install.  So far I’ve only developed one fairly decent print profile which I could easily recreate or retain the file…

 

Worse case scenario, I create new instances of Octoprint on SD cards for the printers and do a fresh install of Cura and Octoprint.  Which might event give me a good excuse to upgrade to RPi 4s…

 

When pasting http://192.168.1.4:api/printer into FireFox, I get a json file that says: “error           "You don't have the permission to access the requested resource. It is either read-protected or not readable by the server."”

 

When I replace “/printer” with “sidewinderX1” I get the same message.

 

I’m totally open to suggestions!

 

By the way, I really do appreciate your help!  I’m not a programmer by any means, nor am I an expert in Linex.  I’ve only been messing with 3D printers for almost 2 years, and I’m still learning stuff!  It’s not easy, but it is fun to learn!

 

Thanks for your help!

fieldOfView, 2022-03-19:
Sorry, I meant http://192.168.1.4:80/api/printer?apikey=your_current_API_key

For some reason, OctoPrint returns a "not found" error on that URL when it is requested by my plugin.

ScottoB1, 2022-03-19:
I uninstalled Cura and then installed the current version.  I’m no longer getting the Not Found error, but I’m still not able to enter the API key.  My current work around is Cura on my iMac.