fieldOfView/Cura-OctoPrintPlugin #269: Blank configuration dialog when pressing "Connect OctoPrint"

nicolas-ivering, 2022-04-22:
Version: 3.7.0-DEV

Cura version: 5 beta

When pressing the button "Connect OctoPrint" in the printers configuration the resulting window is empty. The lines below are from the cura log.

2022-04-22 22:15:04,886 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///C:/Users/power/AppData/Roaming/cura/5.0/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:657:5: Type ManualInstanceDialog unavailable
2022-04-22 22:15:04,886 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///C:/Users/power/AppData/Roaming/cura/5.0/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/ManualInstanceDialog.qml:175:24: RegExpValidator is not a type

fieldOfView, 2022-04-22:
Thanks for the report, I'll get that fixed.

nicolas-ivering, 2022-04-22:
I tried some myself and created https://github.com/fieldOfView/Cura-OctoPrintPlugin/pull/270. There is probably more work to be done but maybe it will save you some typing. At least it works now on my installation.

jneilliii, 2022-04-23:
#270 did resolve the issue of a blank window, but I get the following error when trying to request an API key.

```
Traceback (most recent call last):
  File "C:\Users\jneil\AppData\Roaming\cura\5.0\plugins\OctoPrintPlugin\OctoPrintPlugin\DiscoverOctoPrintAction.py", line 605, in _onRequestFinished
    reply.header(QNetworkRequest.LocationHeader)
AttributeError: type object 'QNetworkRequest' has no attribute 'LocationHeader'
```

fieldOfView, 2022-04-23:
I have pushed a new prerelease of the OctoPrint Connection plugin here:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/releases/tag/v3.7.0-DEV

Should be less prone to crashing, but the print monitor does not seem to work just yet.

jneilliii, 2022-04-23:
confirmed, latest version fixed my error.

Thisismydigitalself, 2022-04-24:
Thank you so much Aldo for your quick fix. OP v3.7.0-DEV_Cura5.0 works for me.
Keep up the good work.

Ingo-Knito, 2022-04-24:
Works great for me too. Thanks for this great work!

Thisismydigitalself, 2022-04-24:
All seems to work OK. i am missing a pop up button to allow me to open my octoprint webpage from within Cura like with 4.13

closed 2022-05-20 by fieldOfView