fieldOfView/Cura-OctoPrintPlugin #271: Prints not starting after upload Cura 5

frankt2012, 2022-04-26:
Prints are uploaded properly to Octoprint but are not staring automatically like they use to in Cura 4.13. Using latest octoprint plugin 3.7.0 on Cura 5.

fieldOfView, 2022-04-27:
I am getting more reports to this effect. I'll look in to it.

frankt2012, 2022-04-27:
> I am getting more reports to this effect. I'll look in to it.

Thank you

fieldOfView, 2022-04-28:
Do you have Print Time Genius installed in OctoPrint?

frankt2012, 2022-04-28:
> I think I have fixed this bug. Could you try the new .curapackage I have created here? https://github.com/fieldOfView/Cura-OctoPrintPlugin/releases/tag/v3.7.0-DEV

I will try it and report back to you. Thank you.

frankt2012, 2022-04-29:
> I think I have fixed this bug. Could you try the new .curapackage I have created here? https://github.com/fieldOfView/Cura-OctoPrintPlugin/releases/tag/v3.7.0-DEV

I can confirm that this has fixed the issue. Actually it looks like everything is working. I am even seeing the video feed and temp readings. Thank you for the fix.

fieldOfView, 2022-04-29:
Awesome, thanks for the quick test!

closed 2022-05-20 by fieldOfView