fieldOfView/Cura-OctoPrintPlugin #272: Plugin not loading on Cura 5 stable release

Charmandrigo, 2022-05-17:
The "connect to octoprint" button is just gone, the plugin is on it's most recent version and does not seem to be loading anymore on Cura 5.0.0, tried to reinstall the plugin several times but nothing works

fieldOfView, 2022-05-17:
The release that is currently in the Marketplace is compatible with Cura 5. I need your cura.log to see what is going wrong on your system. See Help -> Show configuration folder.

Charmandrigo, 2022-05-17:
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/8710891/cura.log)

fieldOfView, 2022-05-17:
It seems like Cura 5 thinks it installed the plugin, but didn't.

Can you tell me what the contents of the `plugins` folder inside the configuration folder looks like? Also please attach the `packages.json` and `plugins.json` files.

Charmandrigo, 2022-05-17:
[json files.zip](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/8711023/json.files.zip)
the folder currently shows only 4 plugins, Cura indeed says I have the octoprint plugin installed 
updating from 4.X to 5.0 broke a lot of plugins and recommended uninstall, it later sync my account and installed some back, but it keeps showing some are broken
[image]

denchuku, 2022-05-18:
@trevorhiley 's solution also worked for me (on Mac OS)

Charmandrigo, 2022-05-18:
> I had the exact same issue. What I ended up doing to fix it was uninstall OctoPrintPlugin and exit Cura 5. Then I got back in and installed OctoPrintPlugin and then again fully exited Cura 5. I then got back into Cura 5 and the plugin folder appeared in the above directory and brought all my settings over and it was working perfectly. Until I followed those exact steps (on Windows) I was having the same problem.

This actually worked! Thanks!

fieldOfView, 2022-05-20:
Did you use the Cura 5.0 beta before? It seems that that version had a bug uninstalling plugins, where it removed the files but not the internal reference that the plugin was installed. So when reinstalling, Cura would think it not necessary to put the files in place because they were already there. 

Let me know if you did not use the Cura 5.0 beta, because then there is an issue in Cura 5 installing plugins.

denchuku, 2022-05-20:
> Did you use the Cura 5.0 beta before? It seems that that version had a bug uninstalling plugins, where it removed the files but not the internal reference that the plugin was installed. So when [...]

I didn't use Cura 5.0 beta before (switched directly from 4 to 5.0.0). Maybe others can confirm that as well?!

fieldOfView, 2022-05-20:
I am reopening this issue for visibility, so I am not inundated by others reporting the same issue.

BaeByte, 2022-05-20:
> I am reopening this issue for visibility, so I am not inundated by others reporting the same issue.

Now I feel really bad about posting that it didn't work last night. I just want you to know we REALLY do appreciate your work thank you so much for everything you do.

fieldOfView, 2022-05-20:
No worries, it is important that these things get reported, so they can get fixed, and so others can find the workaround. I would rather have some double reports than things being broken without me knowing about it. And I may have overreacted with the word "inundated"...

Charmandrigo, 2022-05-20:
I didn't used the betas myself too, I upgraded directly from the version 4 to stable 5 directly, it initally broke several plugins including this one

RyantHults, 2022-05-23:
Just wanted to chime in to say that:

- I had this same issue
- @trevorhiley 's solution worked for me
- I upgraded directly from 4.13.1 to 5.0

telemac73, 2022-05-23:
Same issue.
Upgrade 4.13.1 to 5.0
and i have try to uninstall /reinstall but dont work

StormTrooper, 2022-07-21:
I had the same issue. 
Ubuntu 22.04
Upgraded Cura from 4.13 to 5.10
Had to uninstall plugin, restart cura, install plugin, restart cura. Works after that.

onehunkytenor, 2022-09-10:
> I had the exact same issue. What I ended up doing to fix it was uninstall OctoPrintPlugin and exit Cura 5. Then I got back in and installed OctoPrintPlugin and then again fully exited Cura 5. I then [...]

Worked like a charm for me. Thanks!

herrdeh, 2024-01-02:
ubuntu 22, recent update from CURA 4.11 to 5.65.
Had to remove and reinstall the plugin twice to achieve the "connect octoprint" button.

Anyway - great work, thank you for providing it!

SaschaPeukert, 2024-06-11:
> Long time OctoPrint user here. Just ran into this issue. Happening on a PC that's never run any version of Cura.
> 
> My steps;
> 
> 1. Fresh Install (not an upgrade) of Cura 5.0.0 on Windows 11 PC
> 2. Installed OctoPrint Connection 3.7.2 Plugin from Marketplace
> 3. Restarted Cura
> 4. No sign of OctoPrint anywhere in the Add/Manage Printers UI.  It's just not there, like the plugin isn't even loaded.
> 
> I then found this issue and tried the "fix" steps of uninstall, restart cura, re-install, restart cura.
> 
> No luck.
> 
> I then searched the "plugins" folder in the Cura 5.0 directory tree and there's no sign of OctoPrint Connection.
> 
> I've since tried the "fix" steps several more times, all without success.

I have the exact same issue with Cura 5.7.1. and OctoPrint Connection 3.7.3
All the workarounds mentioned above do not work. No button appears

fieldOfView, 2024-06-11:
I really can't help you without logs.