fieldOfView/Cura-OctoPrintPlugin #275: Unable start print after uploading in cura 5.0

alexyutmp, 2022-05-19:
OctoPrint plugin 3.7.2
Cura 5.0

Error shows after clicking print button
[image]

BaeByte, 2022-05-19:
I can confirm same issue updated to octo 1.8 this afternoon shortly after cura to 5.0 and cannot see connect to octoprint the plugin was also updated from the marketplace 3.7.2

alexyutmp, 2022-05-19:
I got the error like bellow after moonraker and klipper updated to the latest version:
"you are not allowed to upload files to OctoPrint with the configured API key"

fieldOfView, 2022-05-20:
It looks like you are using my plugin with Moonraker instead of OctoPrint. My main concern is supporting OctoPrint, and I think that if Moonraker implements the OctoPrint API, it is their responsibility to make sure their implementation is compatible.

With that said, if you can tell me what my plugin is doing "wrong"  according to Moonraker, eg by inspecting logs created by Moonraker, then I can try to see if I can fix that without breaking things for OctoPrint.

alexyutmp, 2022-05-22:
finally get it work perfectly after recreate an user also updated the key. Sorry for bothering you, close it

closed 2022-05-22 by alexyutmp