fieldOfView/Cura-OctoPrintPlugin #276: When unselecting 'start print after send' you cant toggle the 'switch printer on' checkbox

ImagineerNL, 2022-05-26:
see title

fieldOfView, 2022-06-02:
Please be more descriptive of why this is a problem. If you don't want Cura to start the print, then why would Cura turn on the printer? What is your workflow?

ImagineerNL, 2022-06-02:
My workflow is:
1 slice in cura
2 Uplod to octoprint
3 Turn on printer
4 If needed, preheat and load filament
5 If a while ago since last print, do a bed leveling check and adjust wheels if needed. 
6 Print. 

I know i can technically do 4 and 5 beforehand, but this is a workflow that works for me, also with the least amount of walking for me ((printer with octoprint cam is upstairs)

mvn23, 2022-06-13:
In my setup (Ender 3 running Marlin 2.1), OctoPrint won't let me select a print job unless it is connected to the printer. The printer needs to be switched on before I can connect to it. Having an option to "Select print job after uploading" without an option to power on the printer automatically does not make sense in this case.

closed 2022-06-13 by ImagineerNL

fieldOfView, 2022-06-13:
> Having an option to "Select print job after uploading" without an option to power on the printer automatically does not make sense in this case.

In the next version of the plugin, I'll enable the "Turn on printer with..." control when either "Start print job after uploading" or "Select print job after uploading" is checked.