fieldOfView/Cura-OctoPrintPlugin #277: Memory Leak with Cura 5.x and OctoPrint Plug-in

muckypaws, 2022-05-30:
Hi,

Using the release edition of Cura 5.0 and latest Octoprint Plugin 3.7.2 on macOS 10.15.7, I left a print running on macOS with the Monitor Window open, and left it for several hours.

Cura was unresponsive, iStats was showing over 75Gb memory allocated to Cura.  Raised with Cura team as Issue #12372 but seems the problem is possibly related to the ?plugin
[image]

fieldOfView, 2022-05-30:
Thanks for opening an issue here.
For completeness, here's your original log: https://github.com/Ultimaker/Cura/files/8789491/cura.log.zip

fieldOfView, 2022-05-30:
What is striking me as strange is that the logs show the plugin is using zeroconf version 0.31.0, but it should be using version 0.38.1

```
2022-05-25 15:36:19,837 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevicePlugin.<module> [59]: Using included Zeroconf module version 0.31.0
```

Could you go to Help -> Show configuration folder..., and then zip up at post the contents of the `plugins/OctoPrintPlugin` folder inside the configuration folder that opens?

fieldOfView, 2022-05-30:
Hmm, never mind. I can reproduce the issue that version 0.31.0 is being used instead of the version that should be used. No need to zip up your configuration folder!

fieldOfView, 2022-05-30:
Could you try disabling the UM3 Network Printing plugin (via the Marketplace) and keeping the OctoPrint Connection plugin enabled? It turns out that if the UM3 Network Printing plugin already imports an older zeroconf version, the newer, fixed version that I supply with the plugin is not being used.

muckypaws, 2022-05-30:
Trying this now, it will take several hours for the issue to manifest it self.  Ran for 6 hours this morning and memory usage was at 19Gb, so most likely will update you in the morning.  Thanks for you help so far :)

muckypaws, 2022-05-30:
Hi @fieldOfView 

Disabled the UM3 component and restarted Cura,  Been running two hours and memory already at 8Gb and increasing by around 100mb every second. 

Kind Regards

[image]
[image]

fieldOfView, 2022-05-30:
Could you post a new cura.log?

muckypaws, 2022-05-30:
Here you go... Happy Hunting :) 

[cura.log 2.zip](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/8800131/cura.log.2.zip)
[image]

fieldOfView, 2022-06-02:
Could you test this prerelease of this plugin? https://github.com/fieldOfView/Cura-OctoPrintPlugin/releases/tag/v3.7.3-DEV

* Download the file for your version of Cura (see the "Assets" section).
* Drop the curapackage onto the build volume in Cura, as if you were opening a 3d model.
* Restart Cura.

Please post a new cura.log after you have installed this version of the plugin.

muckypaws, 2022-06-02:
I've just done this, and restarted Cura 5.0, however on the install Plug-In it shows 3.7.2 and not 3.7.3, I'm assuming something more I need do?
[image]
?

fieldOfView, 2022-06-03:
That just shows what version is available in the Marketplace. If you want to see what version is installed, you have to press the cogweheel (top right of the Marketplace dialog). Alternatively, check the top right of the "Connect to OctoPrint" dialog.

muckypaws, 2022-06-03:
Got it, thank you, I have been running your latest code over night.

Observations... Still a memory leak, but not to the extent it was happening before.  Cura was around 650mb at 4pm yesterday, left running overnight (mainly doing not much) and this afternoon (about 22.5 hours later) showing under 8Gb memory usage.

I've attached my latest log.

Kind Regards

[image]
[cura.log 3.zip](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/8832697/cura.log.3.zip)
[image]

muckypaws, 2022-06-03:
I spoke too soon.  Left running and memory over 40Gb now.

[cura.log 4.zip](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/8834716/cura.log.4.zip)
[image]

fieldOfView, 2022-10-24:
I think I have finally fixed loading the newer version of zeroconf here:
b4b7d683f2a3f5b716eb787a2232df69c6dc0145