fieldOfView/Cura-OctoPrintPlugin #278: Default path

tycho94, 2022-06-01:
Hi, 

Currently the plugin uses '.' as a default path. I don't use any folders on my Octoprint and it fails to upload to the '.' folder. 
When I remove . and upload with an empty path option it works. 

Is there a default path option, or is the '.' path supposed to work?

fieldOfView, 2022-06-02:
> Currently the plugin uses '.' as a default path.

Can you elaborate on that? As far as I am aware, the plugin does not have a "default path"; It uses the filename handed to it by Cura. If there's a forward slash in that, then the plugin shows the part before the last forward slash as the path.
 
Perhaps the path you are talking about is inserted by another plugin? If you post a link to your `Cura.log` (see Help -> Show configuration folder) then perhaps I can see what combination of configurations

tycho94, 2022-06-03:
You're right, it was my mistake. I use the Custom Printjob Naming plugin which was adding a ./ as a path. So indeed the plugin was taking the . from the './' in the name as the path which did not exist on Octoprint.

closed 2022-06-03 by tycho94