fieldOfView/Cura-OctoPrintPlugin #279: Octoprint plugin not available for Qidi printers.

corrado33, 2022-06-02:
Hi, I saw someone posted an issue like this before, but never responded. 

I'm using cura with my Qidi X-Plus by downloading the qidi printer definitions here. https://github.com/alkaes/Qidi-Printer-Definitions/releases/download/1.0/Qidi-Printer-Definitions.v1.0.zip

It works great, and qidi print actually worked.... a bit. But now I have octoprint setup and I'm controlling my qidi that way (it's much more stable.) I tried setting up the octoprint plugin to work with it, but the button to "connect to octoprint" never appears. The button appears for other printers.

Is there anything I can do to update the qidi printer files to allow the plugin here to connect to an octoprint instance controlling a qidi? 

Thanks!

fieldOfView, 2022-06-02:
The problem is that the Qidi printer definitions say that these printers do not support printing over a serial connection. The logical conclusion is that Qidi printers don't support OctoPrint. If they do, then the printer definitions should be altered. 

This is the offending line in the printer definition:
https://github.com/alkaes/Qidi-Printer-Definitions/blob/main/definitions/qidi.def.json#L19

closed 2022-06-02 by fieldOfView

corrado33, 2022-06-02:
Changing that line to true worked great. Thanks! :)