fieldOfView/Cura-OctoPrintPlugin #281: Missing right control

FQQD, 2022-06-12:
I dont know why, but I'm missing the right arrow to move the printer. I reinstalled my OS and it worked fine before :/
Cura 4.4.1 running on Feren OS Linux
OctoPrintPlugin v3.7.2
OctoPrint Version 1.7.3
[image]

fieldOfView, 2022-06-12:
This was fixed in Cura 4.6, two years ago; see https://github.com/Ultimaker/Cura/pull/7126

You may want to use the AppImage version of Cura, which is less likely to be affected by issues like this because it includes a known-working combination of dependencies.

fieldOfView, 2022-06-12:
Also a duplicate of https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/149

closed 2022-06-12 by fieldOfView