fieldOfView/Cura-OctoPrintPlugin #286: Enhancement Request: Support for HLS streams in Monitor

marklieberman, 2022-07-07:
OctoPrint itself natively supports HLS streams as a video source. 

Octopi 0.18 is now bundling an experimental HLS streamer.

It can be enabled by editing /boot/octopi.txt and settings camera_streamer=hls. After rebooting, the stream is available at `/hls/stream.m3u8`.

See also: https://community.octoprint.org/t/experimental-hls-support/20336/72

It is great because I can stream high resolution images from the raspicam at a fraction of the bandwidth of mjpeg.

Unfortunately, the stream is not viewable in Cura.

fieldOfView, 2022-10-10:
I would love to support this, but unfortunately Cura does not come with the ability to display videos. Even displaying the current mjpeg stream is a hack; the stream is interpreted as several consecutive still jpeg images, and fed to the window one at a time.

marklieberman, 2022-10-12:
I suspected as much from looking at the plugin source, although I have almost no experience with Python. My idea was to use something like python-vlc to extract frames from the HLS stream and convert them into whatever format is most suitable for display. Documentation is hard to come by, but I found [some encouraging code here](https://github.com/oaubert/python-vlc/issues/17#issuecomment-277476196).

fieldOfView, 2022-10-17:
The problem with python-vlc is that it would rely on an externally installed vlc. I would probably need to keep the current implementation as a fallback, and add instructions to install vlc separately.

Arketsu, 2023-06-01:
@fieldOfView I am not sure is this is completely related or if I should create a brand new ticket but is it possible to add m3u8 stream support to octoprint  plugin? Currently I am running wyze_bridge to get the stream and it works great in the octoprint webpage but no feed in cura.