fieldOfView/Cura-OctoPrintPlugin #289: Segmentation fault OctoPrintPlugin 3.7.2, cura 4.13.1

m42j, 2022-07-23:
I'm on Manjaro/ArchLinux cura version 4.13.1-1

When I start cura with OctoPrint connection 3.7.2 installed I get a segmentaton fault.

Deleting OctoPrintPlugin from plugins directory and cura starts again.

But after again installing OctoPrint connection 3.7.2 cura will not start.

[output.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/9174152/output.log)

fieldOfView, 2022-07-23:
Are you using the AppImage provided by Ultimaker? If not, could you try if the same crash happens with that version?
You can download it here:
https://github.com/Ultimaker/Cura/releases/tag/4.13.1

m42j, 2022-07-23:
Now tested the AppImage and it does not crash.

andyboeh, 2022-07-30:
Same for me, on Arch Linux.

fieldOfView, 2022-07-30:
Then unfortunately I am going to blame this on "uneven" dependencies that are not supported by Ultimaker. Unless someone shows me where I am doing something wrong I am going to have to close this.

closed 2022-07-30 by fieldOfView