fieldOfView/Cura-OctoPrintPlugin #295: OctoPrintPlugin uses PyQT6 if it is installed, regardless of what the rest of Cura is using -- causes crash

jimduchek, 2022-09-10:
Ran into a segfault when loading the plugin, upon importing PyQT6.  Cura 4.13 uses PyQT5 and I think the conflict is what causes the crash. Putting a 'raise ImportError' into the PyQT6 import blocks (forcing it to fall back to PyQT5) 'fixed' the problem.  

Not sure if Cura provides any way to check whether it's using PyQT5 or 6 (I am supposing they switched to 6 in Cura 5?).  Not sure what a proper solution would be.  At the very least, it might be helpful to have a single place to change a flag to force OctoPrintPlugin to use PyQT5.

fieldOfView, 2022-09-18:
I understand how this can be an issue with non-official distributions of Cura. This can be solved by using the official AppImage instead of what your favorite package manager gives you. It may also be worth looking into running Cura from a virtualenv, so you have control over which versions of python modules are used.

I will look in to a better way to switch between PyQt5 and PyQt6 for a future release of the plugin.

fieldOfView, 2022-10-10:
This is fixed in 9c43ea6. I will leave this issue open until a fixed version is available in the marketplace.

KaiMartin, 2023-01-20:
I tried to apply the fix but stumbled on two issues:

1.  The marketplace still delivers a version from before 9c43ea6 . This is probably because this commit came after the release branch, which is marked as default for this github project.
2. I cloned the plugin and switched to the master branch. This prevented the crash. However, the UI reported that "The plugin Cura-Octoprint-plugin could not be loaded". And indeed, the expected octoprint functions are missing.

My system is Debian/testing.

Here is a transcript of the command line notifications:
`
$ cura
/usr/lib/python3/dist-packages/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
  import imp
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-kmk'
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-kmk'
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-kmk'
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
2023-01-20 22:17:25,131 - DEBUG - [MainThread] UM.Application.initialize [161]: Initializing Ultimaker Cura
2023-01-20 22:17:25,131 - DEBUG - [MainThread] UM.Application.initialize [162]: App Version 4.13.0
2023-01-20 22:17:25,131 - DEBUG - [MainThread] UM.Application.initialize [163]: Api Version 7.9.0
2023-01-20 22:17:25,131 - DEBUG - [MainThread] UM.Application.initialize [164]: Build type None
2023-01-20 22:17:25,132 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/bin/share/uranium/plugins must be a folder.
2023-01-20 22:17:25,132 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/bin/share/cura/plugins must be a folder.
2023-01-20 22:17:25,132 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/lib64/uranium must be a folder.
2023-01-20 22:17:25,132 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/lib32/uranium must be a folder.
2023-01-20 22:17:25,132 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/bin/plugins must be a folder.
2023-01-20 22:17:25,132 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/Resources/uranium/plugins must be a folder.
2023-01-20 22:17:25,132 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/Resources/cura/plugins must be a folder.
2023-01-20 22:17:25,132 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [447]: Initializing storage paths
2023-01-20 22:17:25,132 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [457]: Config storage path is /home/kmk/.config/cura/4.13
2023-01-20 22:17:25,132 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [465]: Data storage path is /home/kmk/.local/share/cura/4.13
2023-01-20 22:17:25,132 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [477]: Cache storage path is /home/kmk/.cache/cura/4.13
2023-01-20 22:17:25,132 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/lib/python3/dist-packages/UM/../plugins must be a folder.
2023-01-20 22:17:25,132 - INFO - [MainThread] UM.PackageManager.__init__ [55]: Found bundled packages JSON file: /usr/share/cura/resources/bundled_packages/cura.json
2023-01-20 22:17:25,132 - INFO - [MainThread] UM.PackageManager.__init__ [55]: Found bundled packages JSON file: /usr/share/uranium/resources/bundled_packages/uranium.json
2023-01-20 22:17:25,132 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [123]: Trying OpenGL context 4.1...
2023-01-20 22:17:25,132 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [131]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
2023-01-20 22:17:25,132 - DEBUG - [MainThread] UM.Qt.QtApplication.initialize [177]: Detected most suitable OpenGL context version: 4.1 Core profile
2023-01-20 22:17:25,132 - INFO - [MainThread] UM.Qt.QtApplication.initialize [184]: Initializing job queue ...
2023-01-20 22:17:25,132 - INFO - [MainThread] UM.Qt.QtApplication.initialize [188]: Initializing version upgrade manager ...
2023-01-20 22:17:25,132 - INFO - [MainThread] UM.PackageManager._loadManagementData [170]: Loaded bundled packages data from /usr/share/cura/resources/bundled_packages/cura.json
2023-01-20 22:17:25,132 - INFO - [MainThread] UM.PackageManager._loadManagementData [170]: Loaded bundled packages data from /usr/share/uranium/resources/bundled_packages/uranium.json
2023-01-20 22:17:25,132 - INFO - [MainThread] UM.PackageManager._loadManagementData [197]: Loaded user packages management file from /home/kmk/.local/share/cura/4.13/packages.json
2023-01-20 22:17:25,132 - INFO - [MainThread] UM.PackageManager._saveManagementData [270]: Package management file /home/kmk/.local/share/cura/4.13/packages.json was saved
2023-01-20 22:17:25,132 - INFO - [MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [119]: Loading plugin configuration file '/home/kmk/.config/cura/4.13/plugins.json'
2023-01-20 22:17:25,132 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/lib64/cura must be a folder.
2023-01-20 22:17:25,132 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/lib32/cura must be a folder.
[... cut ...]
2023-01-20 22:18:04,534 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyACM0 if baud rate 9600 works. Retry nr: 0. Wait timeout: 3
2023-01-20 22:18:24,052 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyACM0 if baud rate 115200 works. Retry nr: 1. Wait timeout: 15
2023-01-20 22:18:44,070 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyACM0 if baud rate 250000 works. Retry nr: 1. Wait timeout: 15
`

fieldOfView, 2023-01-21:
Sorry, I normally keep issues open until a new release containing the fix is available in the Marketplace. The fixed version is still awaiting review. The `release` branch normally reflects what is in the Marketplace. The `master` branch is where stuff happens (and sometimes lingers).

I think that when checking out the master plugin repository, you did not check out the submodules. See the readme:

> ...
> When cloning the repository, make sure to use the --recursive flag to include the submodules.
> ...

Note that the master branch has some new submodules, so you may have to do something like `git submodule update` after switching branches.

commit 9c43ea6d (2022-10-10): Improve compatibility with unofficial releases before Cura 5.0

Some versions of Cura tried to use the system-installed PyQt6 when they should still be using PyQt5 (which can be installed in parallel)

Fixes #295
closed 2023-02-27 by fieldOfView