fieldOfView/Cura-OctoPrintPlugin #300: OctoPrint not avaliable

prfiredragon, 2022-12-04:
Desktop - 
*Ubuntu 20.04
* Cura 4.13.2
* OctoPrint Connection 3.7.2

`2022-12-03 22:12:58,044 - DEBUG - [MainThread] OctoPrintPlugin.DiscoverOctoPrintAction.testApiKey [345]: Trying to access OctoPrint instance at http://192.168.1.16:5000/ with the provided API key.
2022-12-03 22:13:18,063 - WARNING - [MainThread] OctoPrintPlugin.DiscoverOctoPrintAction._onRequestFailed [566]: Connection refused or timeout when trying to access OctoPrint at http://192.168.1.16:5000/api/settings
2022-12-03 22:13:18,065 - WARNING - [MainThread] OctoPrintPlugin.DiscoverOctoPrintAction._onRequestFailed [566]: Connection refused or timeout when trying to access OctoPrint at http://192.168.1.16:5000/api/settings`

ODROID-C2 - 
*Ubuntu server 20.04
*OctoPrint: 1.8.6

I Need help to fix this issue. I have removed and instaled, the apps and the plugin. I have tested the connection to the URL http://192.168.1.16:5000/api/settings?apikey= MyKEY

prfiredragon, 2022-12-04:
I install [OctoPrintPlugin_v3.7.3-DEV_Cura4.4-4.13.curapackage](https://github.com/fieldOfView/Cura-OctoPrintPlugin/releases/download/v3.7.3-DEV/OctoPrintPlugin_v3.7.3-DEV_Cura4.4-4.13.curapackage) and the problem continues.

fieldOfView, 2022-12-05:
It looks like your OctoPrint instance takes more than 20 seconds to respond to the /api/settings request. I would say that is a problem that needs to be fixed.

You can try increasing the timeout value here:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/blob/release/DiscoverOctoPrintAction.py#L360

prfiredragon, 2022-12-06:
I Change de value to 60000 and I get another error.
Code: `self._settings_reply, 60000, self._onRequestFailed`

Error: API Key is invalid. You can get the API key through the Octoprint web Page.

On Octoprint Settings -> Applications Keys 
I generate a key for cura.

prfiredragon, 2022-12-06:
I test with a usb live of ubuntu 20.04 on another machine and it work. Also appears the request button. I am going to clean this machine.

prfiredragon, 2022-12-07:
Ok, after all this process i know that the problem was the proxy settings on the /etc/environment I add the settings and the problem return. I have no_proxy configured. I remove the proxy information and works again.

prfiredragon, 2022-12-08:
Not q problem with the proxy. Wrong configuración on the proxy. I have 192.168.1.0/24 on no_proxy and dont work. I add 192.168.1.16 the ip of the device and works.

closed 2022-12-08 by prfiredragon