fieldOfView/Cura-OctoPrintPlugin #301: SerialException: device reports readiness to read but returned no data

SteeveGL, 2022-12-07:
Since I updated to 3.7.2 I got an error. I can reproduce by sending a print or clicking Pre-heat from cura.

Cura log
```
...
2022-12-07 22:20:27,509 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onUploadFinished [1596]: Resource created on OctoPrint instance: https://REDACTED/api/files/local/AC_Phone_Stand03.gcode
2022-12-07 22:20:31,003 - WARNING - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._setOffline [1778]: The printer connected to OctoPrint on Chiron is not operational
```

Octoprint log
```
...
2022-12-07 22:07:46,949 - octoprint.util.pip - INFO - ==> pip ok -> yes
2022-12-07 22:07:47,052 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.10.113
2022-12-07 22:07:47,112 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2022-12-07 22:07:47,214 - octoprint.server.util.flask - INFO - Passively logging in user admin from 192.168.10.113
2022-12-07 22:07:47,215 - octoprint.access.users - INFO - Logged in user: admin
2022-12-07 22:07:48,226 - octoprint.util.pip - INFO - Using "/usr/local/bin/python -m pip" as command to invoke pip
2022-12-07 22:07:48,227 - octoprint.util.pip - INFO - pip installs to /usr/local/lib/python3.8/site-packages (writable -> yes), --user flag needed -> no, virtual env -> no
2022-12-07 22:07:48,228 - octoprint.util.pip - INFO - ==> pip ok -> yes
2022-12-07 22:07:48,269 - octoprint.plugins.softwareupdate - INFO - Minimum free storage across all update relevant locations is 4.1GB. That is considered sufficient for updating.
2022-12-07 22:07:49,079 - octoprint.plugins.orvibos20 - INFO - orvibos20 loaded!
2022-12-07 22:07:49,343 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'base_url': 'https://chiron.print-server.home.utopiklab.net/', 'path': '/', 'query_string': 'l10n=en'}
2022-12-07 22:07:49,454 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2022-12-07 22:07:49,847 - octoprint.server.util.sockjs - INFO - User admin logged in on the socket from client 192.168.10.113
2022-12-07 22:07:53,301 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-12-07 22:07:58,621 - octoprint.server.preemptive_cache - INFO - ... done in 9.28s
2022-12-07 22:08:01,212 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-12-07 22:08:01,214 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2022-12-07 22:08:01,244 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-12-07 22:13:59,337 - octoprint.server.util.flask - INFO - Passively logging in user admin from 192.168.10.113
2022-12-07 22:13:59,337 - octoprint.access.users - INFO - Logged in user: admin
2022-12-07 22:13:59,408 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.10.113
2022-12-07 22:13:59,829 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.10.113
2022-12-07 22:13:59,915 - octoprint.server.util.flask - INFO - Passively logging in user admin from 192.168.10.113
2022-12-07 22:13:59,916 - octoprint.access.users - INFO - Logged in user: admin
2022-12-07 22:14:01,516 - octoprint.server.util.sockjs - INFO - User admin logged in on the socket from client 192.168.10.113
2022-12-07 22:14:04,657 - octoprint.filemanager.analysis - INFO - Starting analysis of local:AC_Phone_Stand03.gcode
2022-12-07 22:14:04,661 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /usr/local/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /octoprint/octoprint/uploads/AC_Phone_Stand03.gcode
2022-12-07 22:14:04,720 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: AC_Phone_Stand03.gcode, owner: _api, user: _api
2022-12-07 22:14:04,747 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2022-12-07 22:14:04,754 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: AC_Phone_Stand03.gcode, owner: _api, user: _api
2022-12-07 22:14:04,812 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-12-07 22:14:04,827 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2022-12-07 22:14:05,322 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/octoprint/util/comm.py", line 3997, in _readline
    ret = self._serial.readline()
  File "/usr/local/lib/python3.8/site-packages/octoprint/util/comm.py", line 6672, in readline
    c = self.read(1)
  File "/usr/local/lib/python3.8/site-packages/serial/serialposix.py", line 595, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-12-07 22:14:05,341 - octoprint.util.comm - ERROR - Please see https://faq.octoprint.org/serialerror for possible reasons of this.
2022-12-07 22:14:05,356 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline after error"
2022-12-07 22:14:05,363 - octoprint.filemanager.analysis - INFO - Starting analysis of local:AC_Phone_Stand03.gcode
2022-12-07 22:14:05,366 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2022-12-07 22:14:05,375 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /usr/local/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /octoprint/octoprint/uploads/AC_Phone_Stand03.gcode
2022-12-07 22:14:24,345 - octoprint.filemanager.analysis - INFO - Analysis of entry local:AC_Phone_Stand03.gcode finished, needed 18.98s
```

fieldOfView, 2022-12-12:
Have you tried saving a gcode file and uploading it to OctoPrint and printing it manually?

This seems to be an issue between OctoPrint and your printer(-firmware), beyond the control of my plugin.

SteeveGL, 2022-12-17:
I been able to reproduce directly from OctoPrint. Issue not related to that plugin.

closed 2022-12-17 by SteeveGL