fieldOfView/Cura-OctoPrintPlugin #304: Octprint connection button not shown

Goodsoldier, 2022-12-27:
Some weeks ago, it was no problem. Even my small daughter used well.
But, today I realized that this plug-in is not working properly.

Octoprint connection button in Cura not shown.

Cura version : 5.2.1
Oct Print version : 1.8.6
System : Ubuntu 20.04

Octprint work status is OK. API setting already done

I attached screenshot

[image]

jtommi, 2022-12-27:
I had the same issue and found the answer here #272

fieldOfView, 2022-12-27:
Please uninstall the plugin. Then restart Cura. Restarting Cura now is important.
Install the plugin again, and restart Cura for a second time. The plugin should then be working.

Note that there will never be a "connection button" in the Marketplace. You should be looking in the Printers pane of the preferences instead.

closed 2022-12-27 by fieldOfView