fieldOfView/Cura-OctoPrintPlugin #305: OctoPrint 3.7.2 and Cura 5.3.0-alpha

daviscabral, 2023-01-02:
I decided to take a stab in the new Cura to test a few settings, but I identified an error when sending the print.

[image]

I know that might be too early to report this kind of problem, but in case that is going on that might be addressed by this side, I appreciate it. Let me know if I can help with that too.

fieldOfView, 2023-01-02:
Thanks for reporting. This is caused by a bug in Cura, see https://github.com/Ultimaker/Cura/issues/13675. It also affects Cura 5.2.1, but reportedly it is fixed for the next version of Cura.

fieldOfView, 2023-01-02:
For now you can work around it by disabling the Ultimaker Format Package plugin on your OctoPrint installation. You will have to restart both OctoPrint and Cura (in that order).

closed 2023-01-02 by fieldOfView