fieldOfView/Cura-OctoPrintPlugin #307: Memory leak while monitoring print via Octoprint

Dr-Gozmon, 2023-02-03:
Cura Version
5.2.1

OctoPrint plugin version

3.7.3

Platform
Windows 10

Printer
Ender 3 Pro or Ender 6

Reproduction steps
Opened Cura
Went to Monitor tab
Switch to other printer (it doesn't matter which of the two is selected to begin with)

Actual results
Memory usage starts to increase by about 5MB per second.
Today it got to 12GB of memory used before I realized and closed Cura.

Expected results
Memory usage should not increase to such a level

Additional information & file uploads
Including cura.log and a screen recording from the most recent time I tested this bug. I didn't capture it very well but the memory usage bounces around just under 400MB normally. It only starts increasing when I switch printers.

https://user-images.githubusercontent.com/123280196/216651940-ea59c2bc-ebd5-4429-8d57-8cf354c33fe1.mp4
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/10580626/cura.log)

fieldOfView, 2023-02-03:
Thanks for transferring the issue. Could you check if the memory leak also happens if you deselect the "Show webcam" checkmark in the Connect to Octoprint dialog, for both printers? Because of the speed at which the memory is leaking, I have a hunch that the camera stream is not properly stopped when switching between printers.

Dr-Gozmon, 2023-02-03:
When I disable the camera stream on both printers, memory usage does not skyrocket like when they are on. Oddly, it does seem to increase memory usage by a couple of MB each time I switch between printers.

fieldOfView, 2023-02-03:
That confirms the camera stream as the first suspect, though there might be a secondary culprit. I'll try to have a look soon.