fieldOfView/Cura-OctoPrintPlugin #310: "Print with OctoPrint" automatically preheats bed but won't preheat extruder

Fraoch, 2023-02-27:
OctoPrint Plugin 3.7.3 on Cura 5.2.2

OctoPrint 1.8.6

Creality Ender 3 V2 Neo

When OctoPrint plugin sends the sliced file to OctoPrint, the bed automatically preheats but the hot end does not preheat. The "Pre-heat" buttons in the Monitor tab are greyed out.

In OctoPrint, the "Tool" temperature is set to "off" and can't be set to anything (just reverts to "off" after setting anything).

After the bed preheats and the print starts, it waits until the hot end reaches the proper temperature, then prints.

It should preheat both the bed and the hot end simultaneously.

fieldOfView, 2023-02-27:
> The "Pre-heat" buttons in the Monitor tab are greyed out.

Before starting a print, or after starting a print? 
The preheat buttons are disabled when starting a print.

> It should preheat both the bed and the hot end simultaneously.

That depends on your start gcode settings.

> In OctoPrint, the "Tool" temperature is set to "off" and can't be set to anything (just reverts to "off" after setting anything).

This plugin cannot fix your OctoPrint instance for you.

Fraoch, 2023-02-27:
My start g-code settings in Cura for my machine don't contain any preheat instructions for either the bed nor the hot end.

They're Cura default, with an added G29 for the ABL:

`; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; ABL
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish`

fieldOfView, 2023-02-27:
If you want them to preheat at the same time, you have to add instructions to the start gcode in such a way that Cura does not add its own preheat sequence.

Add this somewhere in the start gcode:
```
M140 S{material_bed_temperature_layer_0}     ; set bed temperature to e.g. 55 °C and continue
M104 S{material_print_temperature_layer_0}   ; set hot end temperature to e.g. 210 °C and continue
M190 S{material_bed_temperature_layer_0}     ; wait for bed temperature to reach e.g. 55 °C
M109 S{material_print_temperature_layer_0}   ; wait for hot end temperature to reach e.g. 210 °C
```
NB: copy/paste this *as is*, do not change the text between the curly brackets.

Fraoch, 2023-02-27:
Thank you Aldo. I did not realize this was controlled by Cura and not the plugin.