fieldOfView/Cura-OctoPrintPlugin #311: OctoPrint error: OctoPrint responded with an unknown error

ManeLippert, 2023-02-27:
Dear FieldOfView,

I have an Issue for quiet an while on my Macbook Air M2 that can not send data or commands to my OctoPrint Instance.
Under Linux and Windows before the configuration worked without issues in the same network.

I think it is definitely related to #267 because the above error I have had the error message `OctoPrint 404 Not Found Error.`

In `cura.log` the following lines where given:

```
2023-02-27 15:23:02,982 - WARNING - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [1155]: Received an unexpected status code: 400
2023-02-27 15:23:02,987 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [1530]: OctoPrintOutputDevice got an error while accessing http://OCTOPRINT_IP:80/api/printer
2023-02-27 15:23:02,988 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [1535]: OctoPrint responded with an unknown error
```

When I try to use the link [http:///OCTOPRINT_IP:80/api/printer](http:///OCTOPRINT_IP3:80/api/printer) I get this message in Safari:
```
{"error":"You don't have the permission to access the requested resource. It is either read-protected or not readable by the server."}
```

But when I use the following links that you provided in #267:

- [http:///OCTOPRINT_IP:80/api/printer?apikey=API_KEY](http:///OCTOPRINT_IP:80/api/printer?apikey=API_KEY)
   `{"error":"Printer is not operational"}`
- [http:///OCTOPRINT_IP:80/api/job?apikey=API_KEY](http:///OCTOPRINT_IP:80/api/job?apikey=API_KEY)
  `{"job":{"averagePrintTime":null,"estimatedPrintTime":null,"filament":null,"file":{"date":null,"display":null,"name":null,"origin":null,"path":null,"size":null},"lastPrintTime":null,"user":null},"progress":{"completion":null,"filepos":null,"printTime":null,"printTimeLeft":null,"printTimeLeftOrigin":null},"state":"Offline"}`

I get the values of my printer without issues. 

In the Cura Panel for OctoPrint I can connect normal to my OctoPrint instance but than comes the error.

[image]

First I thought this issue is due to the lack of an apple silicon app of Cura and the emulation with Rosetta2 but than a friend of mine who have an Macbook Pro M1 Pro told me he has no issues at all that made me think that the problem is linked to the aggressiv block from web addresses under Mac OS or the Plugin.

Thanks for your help hope we can sort that out.

fieldOfView, 2023-02-27:
> When I try to use the link [http://OCTOPRINT_IP:80/api/printer](http://octoprint_ip3/api/printer) I get this message in Safari

That is because the API key is missing.

> http://OCTOPRINT_IP:80/api/printer?apikey=API_KEY
>   {"error":"Printer is not operational"}

Looks like your printer is turned off and/or not connected to OctoPrint. Do you get the same error when the printer is connected to OctoPrint?

ManeLippert, 2023-02-27:
> That is because the API key is missing.

Ah, I thought so....

> Looks like your printer is turned off and/or not connected to OctoPrint. Do you get the same error when the printer is connected to OctoPrint?

No, when the printer is truned on I get with [http://OCTOPRINT_IP:80/api/printer?apikey=API_KEY](http://octoprint_ip/api/printer?apikey=API_KEY):
```
{"sd":{"ready":true},"state":{"error":"","flags":{"cancelling":false,"closedOrError":false,"error":false,"finishing":false,"operational":true,"paused":false,"pausing":false,"printing":false,"ready":true,"resuming":false,"sdReady":true},"text":"Operational"},"temperature":{"bed":{"actual":17.71,"offset":0,"target":0.0},"tool0":{"actual":16.6,"offset":0,"target":0.0}}}
```

But in `cura.log` I have this message:
```
2023-02-27 16:56:39,471 - WARNING - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [1155]: Received an unexpected status code: 400
2023-02-27 16:56:39,476 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [1530]: OctoPrintOutputDevice got an error while accessing http://192.168.178.93:80/api/printer
2023-02-27 16:56:39,476 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [1535]: OctoPrint responded with an unknown error
```

ManeLippert, 2023-10-26:
Dear FieldOfView,

After testing your Cura Extension with an another OctoPrint instance it worked with some hiccups totally fine, so I think the problem got fixed with the update of Cura to Version 5.4.0.

Thanks for your help!

closed 2023-10-26 by ManeLippert