fieldOfView/Cura-OctoPrintPlugin #313: octoprint cura plugin conflicting with moonraker cura plugin

Arketsu, 2023-04-09:
Currently I have 2 printers.
1. Ender 3 S1 Running with octoprint.
2. Ender 3 V2 running with Klipper , Moonraker.

Recently installed Moonraker plugin to be able to send prints from cura to my klipper printer but found out that the moonraker plugin is conflicting with octoprint plugin. how?

The monitor tab in cura does not show anything if the moonraker plugin is installed when the octoprint plugin is already installed. Only work around I found was to add the url of my moonraker instance and the urls of the camera on my Ender 3 S1 which is the one running octoprint then I can see everything normal in the monitors tab. If I do this another problem pops up which is that cura will default to moonraker and I have to change the destination of the print every time otherwise sends it to the moonraker instance instead of the octoprint.

PS: I will submit the same thing to the moonraker cura plugin github.

fieldOfView, 2023-04-10:
Can you please post a link to your Cura.log? It can be found via Help -> Show configuration folder.

Arketsu, 2023-04-10:
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/11189291/cura.log)

fieldOfView, 2023-04-10:
Nothing jumps out at me immediately from that log, so I will have to try and replicate the issue locally to find out what is going on.

Arketsu, 2023-04-10:
1.Install octoprint plugin 
2.configure octoprint plugin
3.install moonraker plugin
4.restart cura
5.on the monitor tab nothing will show.

6.if you add a random url to the address section and the real url to the camera go to the monitor tab then You will see the camera and controls.

fieldOfView, 2023-04-13:
See https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/issues/70#issuecomment-1507159143

Arketsu, 2023-04-14:
nice find!! exactly what is happening.