fieldOfView/Cura-OctoPrintPlugin #314: Octoprint Connection API Key Issue

tjuoft, 2023-04-13:
I am using Octoprint Connection 3.73.3 with CURA 5.3.0 and Octoprint 1.8.7

When I connect my Creality CR-10 printer in CURA to Octoprint using Octoprint Connection using the Global API Key all functions. 
When I connect my Creality CR-10 printer in CURA to Octoprint using Octoprint Connection using an Application API Key it connects and sends print jobs but monitoring is crippled. Temps do not update. Progress does not display. Pause and Abort Print are greyed out. 

Same printer. Same Octoprint. Same CURA. Only change is the API Key.

fieldOfView, 2023-04-13:
I need to see your cura.log. It can be found via help->show configuration folder.

fieldOfView, 2023-04-25:
A smaller log file would be appreciated. Please also include the octoprint logs from around the same time, so I can see if anything is going wrong on that side.

As far as my plugin is concerned, it does not handle the global API key any differently from the application key. If OctoPrint does not allow the plugin to eg set temperatures, because the application key does not provide enough rights, that should be reflected in the logs.

What I do see in the logs is a couple of instances of "The printer connected to OctoPrint on Pi4B1G is not operational". This generally means that the printer is not connected to the OctoPrint instance. You can still queue (upload) prints then , but not start them, or control the printer.

tjuoft, 2023-04-28:
Well dammit... 
So I deleted the logs
Started a print using global API key, all working well, saved the log. 
I disconnected the printer from Octoprint
I reconnected the printer to Octoprint using an Application key.
Started a print using Application API key and... yup... all working... 

I have no idea what is going on. Octoprint hasn't changed. CURA hasn't changed. 

I even thought "OK, this happened when I first added this printer... lets delete the printer altogether in CURA and start fresh" 
Still works. 

I have no more ideas...

If it happens again I will delete the log, run a print and capture a small log and send it. Until then... closed???

fieldOfView, 2023-04-28:
> Well dammit...

I know that feeling. "Damnit, why can't I get this to _not_ work?". 
Feel free to reopen at any time.

closed 2023-04-28 by fieldOfView