fieldOfView/Cura-OctoPrintPlugin #315: Crashes when i upload a print

mrsamflam, 2023-04-25:
whenever i upload a print cura crashes and gives me an error. i can still control the printer from the "Monitor" tab but not upload prints

Traceback: Traceback (most recent call last):
  File "C:\Users\samga\AppData\Roaming\cura\5.2\plugins\OctoPrintPlugin\OctoPrintPlugin\UploadOptions.py", line 73, in acceptOptionsDialog
    self._proceed_callback()
  File "C:\Users\samga\AppData\Roaming\cura\5.2\plugins\OctoPrintPlugin\OctoPrintPlugin\OctoPrintOutputDevice.py", line 581, in proceedRequestWrite
    if not gcode_writer.write(self._gcode_stream, None):
  File "cura\Utils\Threading.py", line 31, in _call_on_qt_thread_wrapper
    return func(*args, **kwargs)
  File "C:\Program Files\Ultimaker Cura 5.2.1\share\cura\plugins\UFPWriter\UFPWriter.py", line 88, in write
    json.dump(self._getSliceMetadata(), setting_textio, separators=(", ", ": "), indent=4)
  File "json\__init__.py", line 179, in dump
  File "json\encoder.py", line 431, in _iterencode
  File "json\encoder.py", line 405, in _iterencode_dict
  File "json\encoder.py", line 405, in _iterencode_dict
  File "json\encoder.py", line 405, in _iterencode_dict
  File "json\encoder.py", line 438, in _iterencode
  File "json\encoder.py", line 179, in default
TypeError: Object of type SettingFunction is not JSON serializable

mrsamflam, 2023-04-26:
So i guess i missed a cura update so i updated to 5.3.1 and it worked perfectly

closed 2023-04-26 by mrsamflam

fieldOfView, 2023-04-26:
Yes, this was indeed a Cura issue, out of the control of this plugin.