fieldOfView/Cura-OctoPrintPlugin #318: Missing "Connect Octoprint" button on interface

InoSiX, 2023-07-03:
Was just getting around to getting my prints back up and running. I installed the Octoprint connection plugin in Cura but am unable to locate the "Connect Octoprint" button on the interface. I have selected at least 1 printer that has serial connection (for sure, Ender 3), but the button does not exist.

Am I missing something?

closed 2023-07-03 by InoSiX

InoSiX, 2023-07-03:
I restarted multiple times (including PC) to no avail, I attempt to remove and reinstall the plugin. This worked. Not sure what was happening before, but obviously the issue is closed.

fieldOfView, 2023-07-03:
What probably would have helped - but we will never know for sure - is to uninstall the plugin from the Marketplace, then _restart Cura_, then install the plugin again from the Marketplace and finally restart Cura again.