fieldOfView/Cura-OctoPrintPlugin #321: OctoPrint API Key Error

KGuthrie040, 2023-08-17:
Cura Version
5.4.0

Operating System
Windows 11

Printer
Ender 3 Pro with Octoprit

OctoPrint Version
1.92

Every time I try to send a print to OctoPrint I get an API key error.  
[image]

I have tried to reset my API key as well as redownload the addon.  Nothing I have done works.  I have another printer with Octoprint 1.9.0 and I do not have the same issue.  

Has something changed between 1.9.2 and 1.9.0?

robin-thoni, 2024-01-16:
Are you trying to turn on the PSU through OctoPrint with a non-admin account?

[image]

If so, you need to make sure the user you have configured in Cura has the `PSU Control: Control` permission. `Settings` => `Access Control`. Either give your user an additional permission, or add it to a group and give the group the permission.

In my case, I set up a user and added it to the `Operator` group, and gave the `PSU Control: Control` permission to the `Operator` group :

[image]

If not, check your OctoPi logs (`tornado.log`) and search for something like this:
```
2024-01-15 18:28:13 - tornado.access - WARNING - 403 POST /api/plugin/psucontrol (::ffff:10.15.15.127) 44.79ms
```

Routhinator, 2024-12-30:
I'm in the same state as the OP, but no PSU control here. I reinstalled Octoprint with the same hostname. Cura keeps adding the original API key to the API Key field. 

I've deleted the octoprint in Cura, then restarted it, and add it as a new octoprint - but it seems to still pull the exact same _old and invalid_ API key out of it's settings directory somehow, and never takes the new one from the request flow.

I ended up having to add a new printer, then reconnecting Octoprint to get around this problem