fieldOfView/Cura-OctoPrintPlugin #324: Cannot Connect Cura to OctoPrint

Nealtron, 2024-02-15:
I got a new PC and need to install Cura.

- Installed Cura 5.6
- Added my Ender-3 Pro printer profile
- Added Cura-OctoPrintPlugin
- Restarted Cura
- Opened the "Connect to OctoPrint" window
- Added my OctoPrint instance
- Set the API key
- Hit the request button
- The new window opens and I click "Allow" and closed the window.

No matter what, I've tried, it always shows "OctoPrint is not available", and the "Connect" button is always greyed out.

Troubleshooting I've tried:

- Windows firewall is disabled.
- Octoprint is on the same subnet as the PC, no routing or firewall in-between.
- If I click on "Open in browser...", it successfully loads the web interface for my OctoPrint instance.
- If I navigate to "<octoprint-url/api/settings?apikey=<API-Key>", the data loads in successfully in less than 10 seconds. 
- I've attempted to setup the connection with HTTP and HTTPS.
- I've attempted to use FQDN and IP for the Address. 
- I've attempted to use "Application Keys" as well as the "Global API" key. 
- The plugin successfully discovers my octoprint server, and even shows the OctoPrint version, but it still displays "OctoPrint is not available" when configured.
- Added the instance with and without my HTTP username and password.
- Restarting Cura several times.

Any help would be appreciated.

Nealtron, 2024-02-17:
I was able to establish a connection by restarting OctoPrint in Safe Mode.

parchedpacha, 2024-03-04:
I've got the same issue, ultimaker cannot detect my printer. Using the snap package, I was able to inspect the console log, and found that ultimake was trying to connect to an api page that 404'd. maybe there was an API change since the last time this plugin was updated?

closed 2024-03-06 by Nealtron