fieldOfView/Cura-OctoPrintPlugin #326: Wrong Build Plate Temperature

ProtomakerSprint, 2024-03-24:
**Wrong Build Plate Temperature**

I was 3D printing at 60C for the Build plate then I requested 40C for the Build plate

It still 3D Prints ate 60C Build plate

Set to 40C in Cura 3D Print using Octoprint Connection Prints at 60C

This appeared to take place at first: in the very next print, (same file), I lowered the Hot End Temperature from 205C to 195C this also appeared corrected the Build Plate Bed Temperature and not it prints with a Temperature of the asked for 40C. (so it appears that I have to change the Hot End Temperature as well, I may be able to change the hot end back to 205C and still keep 40C on the Build Plate, not tried yet)

**_BUT at the end of the 3D print it displayed 60C again for the build plate_**

Cura 5.7.0. beta 1

G-Code has M190 S40

Octoprint Connection 3.7.3

OctoPrint 1.9.3 Python 3.9.2 OctoPi* 1.0.0cam (build 2023.10.09.154319) Camera Stack

Marlin Marlin 2.1.2.1

Asking for Help
https://community.octoprint.org/t/wrong-build-plate-temperature/57606

fieldOfView, 2024-03-25:
The OctoPrint connection plugin does not change the gcode that Cura generates, so I don't see how this can be an issue with this plugin.

>  then I requested 40C for the Build plate

What exactly do you mean by that? How did you "reuqest" 40C for the build plate? Where did you fill in the number?

ProtomakerSprint, 2024-03-25:
Not OctoPrint: I tried a 3D print directly from an SD Card and get the same issue

Thank You

I set to 40C for the Build Plate and the Hot End to 195C in Cura,

I go to 3D Print and it shows 40C for the Build Plate and Hot End at 195C as I request.

Then just after the Auto Level Bed, just before the 3D Print starts it jumps, ( The display message shows Preheating), to the old setting I used in the past i.e. From requested in Cura 195C Hot End and 40C for the Built Plate to 205C Hot End and 60C Build Plate after which it starts to 3D Print.

closed 2024-03-25 by fieldOfView

ProtomakerSprint, 2024-03-26:
Marlin was

**WAS**

/**
 * Auto-leveling needs preheating
 */
#define PREHEAT_BEFORE_LEVELING // Protomaker Sprint
#if ENABLED(PREHEAT_BEFORE_LEVELING)
  #define LEVELING_NOZZLE_TEMP 210   // Protomaker Sprint (°C) Only applies to E0 at this time
  #define LEVELING_BED_TEMP     60   // Protomaker Sprint
#endif

**_Changed to  i.e. commented out_**
/**
 * Auto-leveling needs preheating
 
#define PREHEAT_BEFORE_LEVELING // Protomaker Sprint
#if ENABLED(PREHEAT_BEFORE_LEVELING)
  #define LEVELING_NOZZLE_TEMP 210   // Protomaker Sprint (°C) Only applies to E0 at this time
  #define LEVELING_BED_TEMP     60   // Protomaker Sprint
#endif
*/