fieldOfView/Cura-OctoPrintPlugin #331: Octoprint Not Available

jadurr, 2024-07-21:
Hello, I have used your plugin before and I just upgraded to Cura 5.7.2 and the plugin does not seem to connect to Octoprint anymore. It now says Octoprint is Not Available after clicking request for the API Key.

dsu-igeek, 2024-12-30:
I'm having the same problem.  It comes and goes, I will see if I can nail down the issue.

younesmaia, 2025-01-29:
Had the same problem on Cura 5.9.0 and was able to make it work by using "octopi.local" instead of the IP address.
This immediately prompted a MacOS dialog stating Cura was requesting access to the local network. After granting permission, not only my manual entry worked ok, but the local OctoPrint instance was automatically detected by the plugin too.
Then all I had to do was to paste the API key and press "request" to connect Cura to the OctoPrint instance. Has been stable ever since.

Therefore, the problem seems related to MacOS security as plugin was not getting the network access prompt up if an IP address is specified. Using a FQDN did the trick.

dsu-igeek, 2025-01-30:
I'm on MacOS and the local network access seems to be causing my problem.  Turning network access on and off for Cura fixes it for a while.

dsu-igeek, 2025-01-30:
Might be caused by https://github.com/Ultimaker/Cura/issues/20076