fieldOfView/Cura-OctoPrintPlugin #332: Enhancement Request: Print with Octoprint or just Upload

SamuelRilling, 2024-11-13:
A simple, but very useful feature i think would be the functionality to select wether printing the sliced model immediatly, or just uploading to the Octoprint server. Right now, if i don't want to print directly, i have 3 choices:
-Save file locally, open up the server and upload manually.
-Start the print , reslice, starting it again and queueing it, then cancelling the print (the ugliest workaround).
-Change the settings to not start print job after uploading, then changing it back again.
While all of those are fine, it would be perfect to have both print and upload as options in the blue print button. For me personally it would make the plugin perfect.

fieldOfView, 2024-11-13:
Enable the "Confirm print job options before sending" option in the Connect to Octoprint dialog, and you will be asked every time if you want to print immediately or not.

SamuelRilling, 2024-11-14:
That is an exact solution! and even already implemented, great work with the plugin and thanks for the incredibly quick reply.

closed 2024-11-14 by SamuelRilling