fieldOfView/Cura-OctoPrintPlugin #333: Print name appears double (.gcode and .ufp)

spanzetta, 2024-12-07:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Slice any 3D file and send it to Octoprint
2. See the name in the popup that appears after have sent the file to Octoprint
See screenshot below

### Actual results

the name appears twice..  as **name**.gode**name**.ufp (see screenshot)

### Expected results

show the name only once (.gode or .ufp)

### Add your .zip and screenshots here ⬇️

[image]

fieldOfView, 2024-12-07:
Thanks for moving the issue here.

Just to be sure, the file is uploaded correctly, right?

spanzetta, 2024-12-08:
> Thanks for moving the issue here.
> 
> Just to be sure, the file is uploaded correctly, right?

yes.. absolutelly yes.. 
It happens with any files.. and I reported it after verified it many times..

wonton1234, 2025-04-19:
same, but it just repeats .gcode twice
for example with file chain.gcode it says "saved to octoprint as chain.gcodechain.gcode"

spanzetta, 2025-08-02:
I continue to see the name as xxxxx.**gcode**xxxxxxx.**ufp**

[image]